losehu/uv-k5-firmware-custom #1: 锁屏时关闭发射

var-tz, 2023-11-30:
哥们你好，非常感谢你汉化的版本，使用起来真的很棒。之前在b站提到的加入mdc1200的问题，没想到也快要实现了。再提个小请求，之前老外那边有个议题，是关于锁屏时关闭发射的，我觉的很有用，有时候在中继守听，难免会误触到ptt，而且k5k6的ptt太容易误触了，就很尴尬。靠菜单里面关闭发射或调整守听方式还是太麻烦，想说话的时候不可能每次都去菜单调整下。麻烦考虑下，感谢 73。祝你生活愉快。

losehu, 2023-11-30:
非常棒的意见，会增加的，我一直以为这个默认就是这样的，太离谱了，居然没有

losehu, 2023-11-30:
我认为既然锁定了按键，就应该禁止任何按键操作，不仅仅是禁止发射，也禁止接受，您觉得如何？

losehu, 2023-11-30:
请尝试这个固件，锁定案件状态下将屏蔽任何按键操作
[firmware.packed.zip](https://github.com/losehu/uv-k5-firmware-chinese/files/13515167/firmware.packed.zip)
非常感谢您的提议，省下了40B的空间

var-tz, 2023-12-01:
> 我认为既然锁定了按键，就应该禁止任何按键操作，不仅仅是禁止发射，也禁止接受，您觉得如何？

不是这个逻辑，正常情况下，按了锁定按钮，其他按键都是不可以操作的，但ptt是可以发射，也可以接收的。(目的是怕按到其他按钮，导致换了频道和频率或其他参数)， 比如我把一个调好的对讲机给队友， 按了f，直接给他用就行了。他也不用担心错误的按到什么，导致双方无法通联。但这时如果我把对讲机放口袋，有可能会误触到ptt键。 如果有个菜单选项可以选择在按f的同时也禁止了tx，那样就不怕误触了。我要说话的时候解锁f的同时也能解锁tx。 这样的话，比按menu，在菜单里面禁止tx方便。(可以选择是否使用)

losehu, 2023-12-01:
我刚测试了官方的固件，他在锁定状态下三个侧键都可以使用，其他的都不可以使用，我会与官方的保持一致

losehu, 2023-12-01:
或者您既然说怕他触碰，还有一种方案是只允许PPT发射与侧键1接收这两个键

losehu, 2023-12-01:
修改为锁定按键时只允许发射、接收按键，与官方中文版本一直