losehu/uv-k5-firmware-custom #13: python打包时生成文件的文件名错误

zhaoparrot, 2023-12-06:
open(sys.argv[3], 'wb').write(packed + digest)这里是不是应该用第4个参数啊？

losehu, 2023-12-06:
你对版本号做了什么修改才导致的错误？

zhaoparrot, 2023-12-06:
我没有修改版本号啊

> arm-none-eabi-objcopy -O binary firmware firmware.bin
> python fw-pack.py **firmware.bin LOSEHU 79e40d9 firmware.packed.bin**
> arm-none-eabi-size firmware
>    text    data     bss     dec     hex filename
>   60092      52    3596   63740    f8fc firmware
这不就应该是第4个参数是 **firmware.packed.bin**么

zhaoparrot, 2023-12-06:
奇怪了  我生成的文件是这样的
[image]

losehu, 2023-12-06:
arm-none-eabi-objcopy -O binary firmware firmware.bin
python fw-pack.py firmware.bin LOSEHU  firmware.packed.bin
arm-none-eabi-size firmware
   text    data     bss     dec     hex filename
  61164      52    3612   64828    fd3c firmware
不知道你的这个那串16进制是哪里来的

zhaoparrot, 2023-12-06:
> arm-none-eabi-objcopy -O binary firmware firmware.bin python fw-pack.py firmware.bin LOSEHU firmware.packed.bin arm-none-eabi-size firmware text data bss dec hex filename 61164 52 3612 64828 fd3c firmware 不知道你的这个那串16进制是哪里来的

VERSION_STRING啊

可能是这里的问题

> # the user might not have/want git installed
> # can set own version string here (max 7 chars)
> ifneq (, $(shell $(WHERE) git))
> 	VERSION_STRING := $(shell git describe --tags --exact-match 2>$(NULL_OUTPUT))
> 	ifeq (, $(VERSION_STRING))
>     	VERSION_STRING := $(shell git rev-parse --short HEAD)
> 	endif
> endif
> #VERSION_STRING := 230930b