losehu/uv-k5-firmware-custom #31: MDC 亚音问题

losehu, 2023-12-09:
通联上下行亚音不同的中继，可以正常解码MDC
通联上下行亚音相同的中继，不能正常解码MDC
直频下无法解码摩托机带亚音的MDC，K5之间可以

losehu, 2023-12-09:
网上传出来的数据手册缺了函数的部分，水平不够，整不出来，摆了
手里也没能发MDC1200的摩托机，不大好试
不来修了，摆了

ETO-bigbang, 2023-12-09:
@oneofeleven help?

losehu, 2023-12-09:
@OneOfEleven
Hello, I have to say that your UVK5 firmware is excellent, especially its MDC1200 signaling. However, during use, I found that it cannot decode the MDC1200 with sub tones sent by Motorola, but it can decode the MDC1200 with sub tones sent by K5.
Thank you very much for your work!

ETO-bigbang, 2023-12-09:
因为发射机型数量所限，目前来看8185，gm300过中继不好使，k5、xts5000好使，我记得gp338也好使，其它机型没有条件测试。

ETO-bigbang, 2023-12-09:
这是一些支持摩托罗拉MDC1200格式的解码和编码的项目，希望有用。
https://github.com/gatekeep/RepeaterController
https://github.com/pabutusa/mdc-decoder
https://github.com/russinnes/fsync-mdc1200-decode
https://github.com/atmatthewat/mdc-encode-decode

ETO-bigbang, 2023-12-09:
2023.12.9-22:00测试，经过中继，中继上行，下行亚音相同，会概率解码，不是完全解不出来！

ETO-bigbang, 2023-12-18:
这是1o11开发MDC的一些说明，也许有用。

https://github.com/OneOfEleven/uv-k5-firmware-custom/issues/173

lloovve, 2024-03-04:
我解决了，注释掉BK4819_send_MDC1200 函数中的BK4819_WriteRegister(0x51, 0);

lloovve, 2024-03-04:
他这个是在发送mdc信令期间关闭的压音，然后中继延时短的话mdc信令传输不完整

losehu, 2024-03-04:
> 我解决了，注释掉BK4819_send_MDC1200 函数中的BK4819_WriteRegister(0x51, 0);

非常感谢，我会尝试

losehu, 2024-03-04:
> 我解决了，注释掉BK4819_send_MDC1200 函数中的BK4819_WriteRegister(0x51, 0);

不是很理解，实测K5在开启接受亚音，摩托机开启同频发射亚音，K5会无法解码摩托机发射的MDC,通过修改BK4819_send_MDC1200函数解决的是什么问题？

losehu, 2024-03-06:
> 我解决了，注释掉BK4819_send_MDC1200 函数中的BK4819_WriteRegister(0x51, 0);

经过测试，并没有用

lloovve, 2024-03-11:
这个函数是发送mdc信令，不是接收信令，如果你原来接收有问题依旧有问题，跟这个修改没有任何关系，这个是K5在发送mdc信令期间不关闭亚音，你要测试信令过中继的时候对方能不能接收修改过的K5发出的mdc信令，这个问题解决的是K5发送出的MDC信令，在过中继传输的时候，对方无法解码，但是直频可以解码的问题

losehu, 2024-03-11:
> 这个函数是发送mdc信令，不是接收信令，如果你原来接收有问题依旧有问题，跟这个修改没有任何关系，这个是K5在发送mdc信令期间不关闭亚音，你要测试信令过中继的时候对方能不能接收修改过的K5发出的mdc信令，这个问题解决的是K5发送出的MDC信令，在过中继传输的时候，对方无法解码，但是直频可以解码的问题

K5发射到中继，用摩托罗拉一直能解啊？

losehu, 2024-03-11:
> 这个函数是发送mdc信令，不是接收信令，如果你原来接收有问题依旧有问题，跟这个修改没有任何关系，这个是K5在发送mdc信令期间不关闭亚音，你要测试信令过中继的时候对方能不能接收修改过的K5发出的mdc信令，这个问题解决的是K5发送出的MDC信令，在过中继传输的时候，对方无法解码，但是直频可以解码的问题

我们上面说的全部都是解码问题，发射MDC没有问题啊？

NICEEEMA, 2024-03-13:
你好大佬，我的台子是1M扩容版K6，有几个问题想请教！
Q1:119K版本打开了首尾音但是发射时在本机上依然无法听到声音。
Q2:短信的中文输入法请问怎么开启？
Q3：DTMF ID是只可以输入ABCD四个字母和数字吗？如果还可以输入其他字母，请问怎么切换输入？

期待您的回答！谢谢！

nyanchan-caicai, 2024-04-03:
不使用88.5亚音的中继 有概率可以通过上下行亚音一样的中继解码MDC

losehu, 2024-04-04:
> 例如我这里本地中继110.9有时可以解码MDC

不是这个原因

ETO-bigbang, 2025-04-05:
opengd77最新的汉化固件，可以很好的解码MDC1200信令，过中继也可以正常发射和解码，不知是否有时间参考一下该固件代码，完善一下k5、k6的信令？

losehu, 2025-04-08:
> opengd77最新的汉化固件，可以很好的解码MDC1200信令，过中继也可以正常发射和解码，不知是否有时间参考一下该固件代码，完善一下k5、k6的信令？

完全没时间