losehu/uv-k5-firmware-custom #34: 当选择ENABLE_SMALL_BOLD := 0编译时报错

ETO-bigbang, 2023-12-11:
ui/main.c: In function 'UI_DisplayMain':
ui/main.c:760:18: error: implicit declaration of function 'UI_PrintStringSmallBold'; did you mean 'UI_PrintStringSmall'? [-Werror=implicit-function-declaration]
  760 |                  UI_PrintStringSmallBold(String, 4, 0, 3);
      |                  ^~~~~~~~~~~~~~~~~~~~~~~
      |                  UI_PrintStringSmall
cc1.exe: all warnings being treated as errors

losehu, 2023-12-11:
下次看看，不过这个字体我认为是必须的
而且我现在已经把字库写到外部eeprom了，节省了3K flash也不会占用内部的flash

ETO-bigbang, 2023-12-11:
不选用这个字体第一是为了节省空间，第二是为了界面协调，字体统一。

你这个改造有点牛啊，常用中文不是梦啊，加油

ETO-bigbang, 2023-12-11:
能把汉字写入eeprom是很棒，但是要牺牲时间，并且很多人还不会刷。

为什么不把不必要的字体，例如ENABLE_SMALL_BOLD选择0来停用，把现有的汉字字体变大一些呢12*12或者16*16？

我这么说并不是说3500个汉字不重要，只是个人的想法，程序可以写的灵活一些，这也是有可以选择的makefile的的原因。

如果你需要3500的汉字，可以选择all_chinese=1把字体启用，再刷入eeprom。
如果不需要可以选择all_chinese=0不启用，这样也能再基本菜单上获得很好的汉字显示。

ETO-bigbang, 2023-12-11:
3500的基本字库还是可以进一步缩减的，既可以获得大字体，也可以缩减时间，字体库也不需要定制。

losehu, 2023-12-11:
我并不会更新全中文字库，因为这很蠢

closed 2023-12-11 by losehu

ETO-bigbang, 2023-12-11:
> 我并不会更新全中文字库，因为这很蠢

全中文字库如果可以一劳永逸，也是一件很酷的事，我认为很多人都会愿意花这一个多小时的时间

losehu, 2023-12-11:
是很6，但是对我来说工作量太大了

ETO-bigbang, 2023-12-11:
> 是很6，但是对我来说工作量太大了

是的，会花很多很多时间，只是调试一个步骤的刷字库就要很久，我修改了main.c的mdc的UI字体显示，这样就可以停用粗字体了，又生了0.6k空间，如果能用这0.6k把汉字变大就好了，可是我还不会，你有汉字的相关说明吗？我想学一下。

我看main.c中大部分显示都是UI_PrintStringSmall

只有3处用了粗字体，并做了判断，如果可以的话，可以再mdc中加个判断吗？谢谢，这样我就不用每次改main.c了，谢谢
#ifdef ENABLE_SMALL_BOLD
                    UI_PrintStringSmallBold("TX", 14, 0, line);
#else
                    UI_PrintStringSmall("TX", 14, 0, line);
#endif

losehu, 2023-12-11:
我其实更喜欢粗字体，所以我不会去改
主要是太麻烦了，不想整，其实现在已经可以用12*12的字体，但我太懒了

ETO-bigbang, 2023-12-11:
尊重你

losehu, 2023-12-11:
事实证明我是一个傻逼，我把64Kb看成了64KB，也就是说eeprom其实只剩下300多B，而联系人占用了176B，实际上，只剩下100多B的eeprom了

ETO-bigbang, 2023-12-11:
> 事实证明我是一个傻逼，我把64Kb看成了64KB，也就是说eeprom其实只剩下300多B，而联系人占用了176B，实际上，只剩下100多B的eeprom了

想变大字体，关闭粗字体才是最好的选择