losehu/uv-k5-firmware-custom #127: S-meter don't match RSSI

UID163844189, 2024-06-15:
At frequency over 30MHz, S9 should match -93dBm. But at the frequency of 438.5 MHz, displays S0 with RSSI-127 dBm. 
firmware: LOSEHU126.bin

losehu, 2024-06-15:
Actually, I am not very familiar with this part. Due to the lack of instruments and equipment, I can only follow the firmware settings of Egzumer. Have you tried using his firmware for measurements?

UID163844189, 2024-06-15:
I tried Egzumer which displays S4 with RSSI -102. This is also wrong. -102dBm should match S7+3. 
I have chance to use radio test set. I can help you to test if these firmwares has a accurate RSSI meter (or they have a same offset).

lxh1099, 2024-06-16:
大神 您好！  刷了 Luosehu固件后，在普通频率或信道模式下，用信号源测试，K5 K6的静噪灵敏度门槛大幅度下降，大概在-116db,   比官方原生固件的 -123 下降了很多，也就意味着作为对讲机最最重要的一个参数 灵敏度 下降了大概4倍左右。奇怪的是很少见有人在论坛上讨论这个事情。  希望大神能看到这个问题，能解决静噪灵敏度下降 以及 S场强显示 和 RSSI   db数显示 不准的问题。

UID163844189, 2024-06-16:
-123的灵敏度是来自于bk4819的数据手册，也就是这帮泉州机使用的fm对讲机集成芯片。机器标称的灵敏度我不清楚是实测为-123还是只是直接把手册上的-123拿过来了。查看图纸，射频信号转为音频信号的处理是不经过mcu的。
音频信号从bk4819出来后，与bk1080收音机芯片，h3601语音芯片的信号合并，进入ln4898功放，而功放从图纸上看是与mcu有连接的。
可以检查功放的工作状态与原版固件是否一致。

losehu, 2024-06-26:
> 大神 您好！ 刷了 Luosehu固件后，在普通频率或信道模式下，用信号源测试，K5 K6的静噪灵敏度门槛大幅度下降，大概在-116db, 比官方原生固件的 -123 下降了很多，也就意味着作为对讲机最最重要的一个参数 灵敏度 下降了大概4倍左右。奇怪的是很少见有人在论坛上讨论这个事情。 希望大神能看到这个问题，能解决静噪灵敏度下降 以及 S场强显示 和 RSSI db数显示 不准的问题。

这是因为使用iic从eeprom读取了过多数据干扰电路，实测不用中文信道就能解决这个问题

f3n9, 2024-06-27:
> > 大神 您好！ 刷了 Luosehu固件后，在普通频率或信道模式下，用信号源测试，K5 K6的静噪灵敏度门槛大幅度下降，大概在-116db, 比官方原生固件的 -123 下降了很多，也就意味着作为对讲机最最重要的一个参数 灵敏度 下降了大概4倍左右。奇怪的是很少见有人在论坛上讨论这个事情。 希望大神能看到这个问题，能解决静噪灵敏度下降 以及 S场强显示 和 RSSI db数显示 不准的问题。
> 
> 这是因为使用iic从eeprom读取了过多数据干扰电路，实测不用中文信道就能解决这个问题

您好，您提到的“不用中文信道”，是指不用中文命名信道吗？还是不用支持“中文信道”的固件版本？谢谢！

losehu, 2024-07-05:
这个固件如何呢？
https://github.com/DualTachyon/uv-k5-firmware

losehu, 2024-07-05:
> > > 大神 您好！ 刷了 Luosehu固件后，在普通频率或信道模式下，用信号源测试，K5 K6的静噪灵敏度门槛大幅度下降，大概在-116db, 比官方原生固件的 -123 下降了很多，也就意味着作为对讲机最最重要的一个参数 灵敏度 下降了大概4倍左右。奇怪的是很少见有人在论坛上讨论这个事情。 希望大神能看到这个问题，能解决静噪灵敏度下降 以及 S场强显示 和 RSSI db数显示 不准的问题。
> > 
> > 
> > 这是因为使用iic从eeprom读取了过多数据干扰电路，实测不用中文信道就能解决这个问题
> 
> 您好，您提到的“不用中文信道”，是指不用中文命名信道吗？还是不用支持“中文信道”的固件版本？谢谢！
不用支持“中文信道”的固件版本

IvanMaestroGTR, 2024-07-06:
> At frequency over 30MHz, S9 should match -93dBm. But at the frequency of 438.5 MHz, displays S0 with RSSI-127 dBm. firmware: LOSEHU126.bin

Actually you just have to set the s meter in egzumer chirp, set the s9 threshold as -92 and set s0 threshold to -140

You will have the s meter that is compliant to the ITU standards 😁

losehu, 2024-09-23:
允许用户自行调节，可以参考Egzumer固件的调节方式

closed 2024-09-23 by losehu

IvanMaestroGTR, 2025-04-07:
I believe you might have to use the older egzumer module, the later version don't support the heavily modified losehu firmware