losehu/uv-k5-firmware-custom #135: 非常抱歉，再次打扰您（上次我非常抱歉），这次希望我没有出错，compile-with-docker.sh这个脚本我试过发生一些错误

xingzitian, 2024-07-02:
根据我的一些发现，这个是由windows的处理脚本直接搬过来的，在其他作者代码中此文件有所不同，麻烦您看一下是不是可以改为：`docker run --rm -v "$(pwd)/compiled-firmware:/app/compiled-firmware" uvk5 /bin/bash -c "cd /app && rm -rf compiled && make clean && make && cp *.bin compiled-firmware/"`
以下是这个代码运行的结果

>docker run --rm -v "$(pwd)/compiled-firmware:/app/compiled-firmware" uvk5 /bin/bash -c "cd /app && rm -rf compiled && make clean && make && cp *.bin compiled-firmware/"
chmod +x del_linux.sh
sh ./del_linux.sh
Deletion complete.
rm -f *.bin
make build ENABLE_CHINESE_FULL=0 ENABLE_ENGLISH=1 ENABLE_FMRADIO=1 ENABLE_MESSENGER=1 ENABLE_MESSENGER_DELIVERY_NOTIFICATION=1 ENABLE_MESSENGER_NOTIFICATION=1 ENABLE_SPECTRUM=1 ENABLE_MDC1200=1 ENABLE_MDC1200_EDIT=1 ENABLE_MDC1200_CONTACT=1
make[1]: Entering directory '/app'
E
chmod +x del_linux.sh
sh ./del_linux.sh
Deletion complete.
python fw-pack.py firmware.bin LOSEHU LOSEHU130E.bin
arm-none-eabi-size firmware
   text    data     bss     dec     hex filename
  61032      52    3532   64616    fc68 firmware
make[1]: Leaving directory '/app'
make build ENABLE_CHINESE_FULL=4 ENABLE_ENGLISH=1 ENABLE_DOPPLER=1 ENABLE_SPECTRUM=1 ENABLE_FMRADIO=1 ENABLE_MDC1200=1 ENABLE_MDC1200_EDIT=1 ENABLE_MDC1200_CONTACT=1
make[1]: Entering directory '/app'
EK
chmod +x del_linux.sh
sh ./del_linux.sh
Deletion complete.
python fw-pack.py firmware.bin LOSEHU LOSEHU130EK.bin
arm-none-eabi-size firmware
   text    data     bss     dec     hex filename
  61268      52    3596   64916    fd94 firmware
make[1]: Leaving directory '/app'
make build ENABLE_CHINESE_FULL=0 ENABLE_SPECTRUM=1 ENABLE_FMRADIO=1 ENABLE_MDC1200=1 ENABLE_MDC1200_EDIT=1 ENABLE_MDC1200_CONTACT=1
make[1]: Entering directory '/app'
chmod +x del_linux.sh
sh ./del_linux.sh
Deletion complete.
python fw-pack.py firmware.bin LOSEHU LOSEHU130.bin
arm-none-eabi-size firmware
   text    data     bss     dec     hex filename
  61272      52    3276   64600    fc58 firmware
make[1]: Leaving directory '/app'
make build ENABLE_CHINESE_FULL=4 ENABLE_DOPPLER=1 ENABLE_SPECTRUM=1 ENABLE_FMRADIO=1 ENABLE_MDC1200=1 ENABLE_MDC1200_EDIT=1 ENABLE_MDC1200_CONTACT=1
make[1]: Entering directory '/app'
K
chmod +x del_linux.sh
sh ./del_linux.sh
Deletion complete.
python fw-pack.py firmware.bin LOSEHU LOSEHU130K.bin
arm-none-eabi-size firmware
   text    data     bss     dec     hex filename
  61336      52    3600   64988    fddc firmware
make[1]: Leaving directory '/app'
make build ENABLE_CHINESE_FULL=4 ENABLE_DOPPLER=1 ENABLE_PINYIN=1 ENABLE_SPECTRUM=1 ENABLE_FMRADIO=1
make[1]: Entering directory '/app'
H
chmod +x del_linux.sh
sh ./del_linux.sh
Deletion complete.
python fw-pack.py firmware.bin LOSEHU LOSEHU130H.bin
arm-none-eabi-size firmware
   text    data     bss     dec     hex filename
  61080      56    3528   64664    fc98 firmware
make[1]: Leaving directory '/app'
make build ENABLE_CHINESE_FULL=4 ENABLE_PINYIN=1 ENABLE_4732=1 ENABLE_4732SSB=1  ENABLE_SPECTRUM=1
make[1]: Entering directory '/app'
H
SI4732
chmod +x del_linux.sh
sh ./del_linux.sh
Deletion complete.
python fw-pack.py firmware.bin LOSEHU LOSEHU130HS.bin
arm-none-eabi-size firmware
   text    data     bss     dec     hex filename
  61156     188    3188   64532    fc14 firmware
make[1]: Leaving directory '/app'
[image]

losehu, 2024-07-05:
想看一下用我的sh报什么错

losehu, 2024-07-05:
你是正确的，类unix应该用pwd而不是CD

closed 2024-07-05 by losehu