losehu/uv-k5-firmware-custom #145: K5WEB - Firmware Not Supported when writing SI4732 SSB Patch - LOSEHU130ES

jeffs555, 2024-08-09:
On K5WEB I went to LOSEHU DIY and loaded firmware with English, spectrum, SI4732 and enable SSB. The firmware displays LOSEHU130ES. When I tried to write the SSB Patch, K5WEB says Firmware Not Supported. I got it to work by loading LOSEHU130HS firmware, then writing the SSB Patch, then reloading the LOSEHU130ES.

losehu, 2024-08-10:
well,that's a problem

losehu, 2024-08-11:
Hello, this issue is not due to a firmware bug but rather because the multiple versions are difficult to manage. Only the firmware versions with 'K' and 'H' are for machines that have already been equipped with a larger EEPROM by default. Due to the difficulty in managing the numerous firmware versions, we recommend using K5WEB for customizing features.
https://k5.vicicode.com/#/idea/losehu

closed 2024-08-11 by losehu