losehu/uv-k5-firmware-custom #146: Add support for higher capacity battries

OldMateAidos, 2024-08-11:
Iam loving this firmware so so much the abilities in this firmware are very impressive , is there anyway to implement feature for higher capacity battries

losehu, 2024-08-11:
* fork the code via github

* change gSubMenu_BATTYP in "ui/menu.c"

* change  ` case MENU_BATCAL:
            *pMin = 1600;
            *pMax = 2200;
            break;` in app/menu.c
            
* then rebuild the firmware via github action

closed 2024-08-11 by losehu

OldMateAidos, 2024-08-11:
Absolute legend , thank you

OldMateAidos, 2024-08-12:
How do I add boot image support instead of boot text , I have managed to implement larger battery , what array do I have to change to allow three battery options in the sub battery menu ?

losehu, 2024-08-13:
How do I add boot image support instead of boot text , I have managed to implement larger battery , what array do I have to change to allow three battery options in the sub battery menu ?

翻译成中文

losehu, 2024-08-13:
only "EK" could use boot image,write image in  https://k5.vicicode.com/#/idea/Image
as for "three battery options"，This is a bit troublesome. If it's not necessary, two would be enough

OldMateAidos, 2024-08-13:
Thank you , trying to implant your auto Doppler shift and SI4732 support into a firmware im working on for just scanning and weather data and sat images data could you give me some guidance?, if I replace eprom do I just have to flash again or further steps need , very thankfull for your help

losehu, 2024-08-13:
ok，but how can 4732 used to NOAA
OTHERWISE,Doppler shift may can't do that

OldMateAidos, 2024-08-13:
I trying to also add 4732 support as well for higher hf , and use normal VHF for NOAA I use  radio to pass audio to PC get weather image , is it not possible to use both ?

losehu, 2024-08-13:
The frequency calculation of Doppler is entirely done by a computer, and if you are the same, you should be able to do so