losehu/uv-k5-firmware-custom #165: 打包固件问题

Cao150702, 2024-09-22:
[image]

Cao150702, 2024-09-22:
[image]

losehu, 2024-09-22:
图片没传上来

Cao150702, 2024-09-22:
我去秒回  图片传不上去了哈哈哈哈 运行sh文件后 
@Cao150702 ➜ /workspaces/uv-k5-firmware-custom (main) $ '/workspaces/uv-k5-firmware-custom/compile-with-docker.sh'
bash: /workspaces/uv-k5-firmware-custom/compile-with-docker.sh: Permission denied

losehu, 2024-09-22:
管理员运行

Cao150702, 2024-09-22:
好像还是不行  是方法不对吗 sudo su
root ➜ /workspaces/uv-k5-firmware-custom (main) $ '/workspaces/uv-k5-firmware-custom/compile-with-docker.sh'
bash: /workspaces/uv-k5-firmware-custom/compile-with-docker.sh: Permission denied

Cao150702, 2024-09-22:
用系统终端也尝试了一下 
mr.cao@192 ~ % sudo -i
Password:
192:~ root# /Volumes/个人文件/uv-k5-firmware-custom-main/compile-with-docker.sh -sh: /Volumes/个人文件/uv-k5-firmware-custom-main/compile-with-docker.sh: Permission denied
192:~ root#

losehu, 2024-09-22:
chmod

closed 2024-09-22 by losehu