losehu/uv-k5-firmware-custom #181: 是否能添加：本固件屏幕显示内容含义的说明文档

00GITHUBUSER00, 2024-10-18:
例如左上角的PS，R，S,  DWR，-87，S7各自是什么含义的说明文档

nanhantianyi, 2024-10-19:
状态栏：
`PS` = power save (省电模式)
`T` = `TX` = transmit (信号发送中)
`R` = `RX` = receive (信号接收中)
`S `= scan (频率扫描中)
`*` = 扫描所有信道
`1` = 扫描信道列表1
`2` = 扫描信道列表2
`DWR` = dual watch response (双频守候双频响应)
`DW` = dual watch (主信道发射，双信道接收)
`XB` = cross band (主信道发射，副信道接收)
`><` = dw on hold (双守时锁定当前活动信道)

中间栏：
`-87` = -87 dbm (信号强度)
`S7` = 场强表，相当于信号强度分级，数值为1-9，如果比9级还强的信号 显示为`+10`这种样式， 表示 S9 + 10 

如果还有不明白的，请列出来，这里是 **_BI1UTY_**，73！

00GITHUBUSER00, 2024-10-19:
1.在以下的各种通信模式中，日常的语音通联属于哪一种模式，是如何界定出来的？
2.RST信号评价时，59是个人判断的可读性+直接使用场强表的数值么？
3.例如在中继台频率439.750 MHz上进行语音通联属于哪种模式，怎么界定是这个模式的？
4.泉盛K6低中高三种发射功率分别对应多少瓦呀，其他手台也是这样划分的么？
5.为什么听广播的的时候播报的是88.0，但是我在88.0听不到，98.4听到了但是质量不太好？

这里是BI3TAO，感谢您的回答

### 1. **NFM（Narrowband Frequency Modulation）**
   - 窄带频率调制，FM的一个变种，带宽较窄（一般在12.5kHz或更小）。
   - 常用于对带宽要求高的通信，如对讲机、航空通信和应急频段。

### 2. **WFM（Wideband Frequency Modulation）**
   - 宽带频率调制，用于传输高质量音频或复杂信号，带宽较宽（一般88-108 MHz之间的FM广播）。
   - 常见于FM广播电台和电视音频传输。

### 3. **LSB（Lower Sideband）**
   - **下边带调制**，在单边带（SSB）系统中使用。调制信号中只保留载波以下频率的部分。
   - 多用于业余无线电通信和短波段通信，效率高且节省带宽。

### 4. **USB（Upper Sideband）**
   - **上边带调制**，与LSB相反，只保留载波以上频率的部分。
   - 通常用于业余无线电通信的高频段，以及短波通信的语音传输。

### 5. **SSB（Single Sideband）**
   - **单边带调制**，是一种AM的变种，只传输上边带或下边带，从而节省带宽和功率。
   - 广泛用于军事、航空和海事通信。

### 6. **CW（Continuous Wave）**
   - **连续波通信**，通常指莫尔斯电码。用载波的开关（开/关状态）来传递信息。
   - 常用于业余无线电和海上通信，因其高可靠性和低带宽需求。

### 7. **AM（Amplitude Modulation）**
   - **幅度调制**，通过改变载波信号的幅度来传递信息。
   - 应用于中波广播、航空和短波通信。缺点是易受干扰。

### 8. **FM（Frequency Modulation）**
   - **频率调制**，通过改变载波的频率来传输信号。相比AM更抗干扰。
   - 常见于FM广播电台和对讲机。

### 9. **FSK（Frequency-Shift Keying）**
   - **频移键控**，数字调制技术，信息用两个或多个频率的切换表示。
   - 用于低速数据传输，如无线电信标、远程控制。

### 10. **RTTY（Radio Teletype）**
   - **无线电传字机**，一种数字通信模式，通过无线电波传输文本数据。
   - 常用于气象、航空和业余无线电数据传输。

### 11. **SSTV（Slow-Scan Television）**
   - **慢扫描电视**，一种图像传输方式，将图片通过无线电传送，每秒传输的帧数很低。
   - 常用于业余无线电操作，用于传送静态图片和实验性通信。

---

### 2. **RST 评价系统**  

**RST** 是一种用于无线电通信中的信号质量报告系统，评估无线信号的可读性、强度和音调（如果是电码通信）。它有三部分组成：

- **R**（Readability）：信号的可读性，用 1 到 5 评分。
- **S**（Signal Strength）：信号强度，用 1 到 9 评分。
- **T**（Tone）：音调清晰度，仅用于电码通信（CW），评分为 1 到 9。

#### **评分标准：**
- **R（可读性）**：
  - 1：完全不可读
  - 2：断断续续，有时能理解
  - 3：勉强可读，但困难
  - 4：基本清晰，容易理解
  - 5：完全清晰可读

- **S（信号强度）**：
  - 1：极弱，几乎无法检测
  - 2：非常弱
  - 3：弱但可接收
  - 4：信号较弱
  - 5：信号相当清晰
  - 6：信号良好
  - 7：信号非常强
  - 8：强信号
  - 9：极强信号

- **T（音调）**（仅限电码通信）：
  - 1：非常粗糙的音调，无法辨识
  - 2-4：音调略有瑕疵，音质不纯
  - 5-6：音调尚可接受，略微不纯
  - 7-8：音调清晰，有轻微瑕疵
  - 9：完全纯净音调

#### **示例：**
- **59**：清晰且强的语音信号（语音通信常省略T部分）。
- **579**：CW 通信中信号完全清晰且音调纯净。
- **229**：信号非常弱，但音调纯净。

#### **在 439.750 MHz 上的 RST 使用：**
- 如果通过此频率进行语音通信，一般会用 **RS** 进行评价（例如“59”）。
- 如果是电码通信（不常见于此频率），则使用 **RST**，例如“579”。

---

### 3. **439.750 MHz 频率的用途**  
439.750 MHz 属于 **超高频 (UHF) 波段**，这是在 **430-440 MHz** 范围内的一部分。该频段主要用于：

- **业余无线电**（Amateur Radio）  
  - UHF 波段常用于 **中继台 (Repeater)**，增强信号覆盖范围。
  - 439.750 MHz 很可能是 **业余无线电中继台的接收频率**，用于语音通信。中继台接收来自电台的信号后，通过另一频率转发，以扩大通信范围。

- **NFM（Narrowband Frequency Modulation）**  
  该频段常采用 NFM（窄带频率调制），确保在密集的频率分配中减少干扰。NFM 主要用于业余爱好者之间的语音通信以及应急通信。

nanhantianyi, 2024-10-19:
1、业余无线电UV段使用的是FM调制进行语音通联，使用何种调制模式跟波长、带宽、传输效率、使用环境等多种因素有关。
2、语音通联信号报告提供RS即可，R为话音可辨程度，1-5按感受给即可，S为场强，报场强表读数。
3、同第一点
4、大概是1、3、5W吧，我也没测过具体数值。每个品牌都不一样。
5、FM广播的频率虽然和你频率模式下设置的频率一样，但是带宽不同，FM广播的带宽更大，所以会无法收听或收听失真严重

我也不是专业的，如有错误请指正，这里是 **_BI1UTY_**，73！

00GITHUBUSER00, 2024-10-21:
这里是BI3TAO，感谢 BI1UTY的回答，73。

closed 2024-10-21 by 00GITHUBUSER00