losehu/uv-k5-firmware-custom #207: 利用抽象方法减少Flash占用的方法

xiongsp, 2025-05-19:
你好，很感谢你改写的开源固件。目前固件体积占用过大，无法实现”全功能“，比如`LOSEHUxxxH`版本不能加入短信功能。

为了解决占用过大问题，我在以下环境中进行了测试。实验表明，仅仅针对一个函数的优化即可实现约168字节的空间释放。（类似可优化的函数有许多，预计能释放出1KB，仅估算未实测）

原理：bk4819.c存在大量重复`BK4819_WriteRegister`/`BK4819_ReadRegister`函数调用，使用循环可大大降低代码量；

DEMO：
函数`BK4819_InitAGC`，所注释部分为原实现
```c
void BK4819_InitAGC(bool amModulation) {
    // REG_10, REG_11, REG_12 REG_13, REG_14
    //
    // Rx AGC Gain Table[]. (Index Max->Min is 3,2,1,0,-1)
    //
    // <15:10> ???
    //
    // <9:8>   LNA Gain Short
    //         3 =   0dB  <<<		1o11				read from spectrum			reference manual
    //         2 = 					-24dB  				-19     					 -11
    //         1 = 					-30dB  				-24     					 -16
    //         0 = 					-33dB  				-28     					 -19
    //
    // <7:5>   LNA Gain
    //         7 =   0dB
    //         6 =  -2dB
    //         5 =  -4dB
    //         4 =  -6dB
    //         3 =  -9dB
    //         2 = -14dB <<<
    //         1 = -19dB
    //         0 = -24dB
    //
    // <4:3>   MIXER Gain
    //         3 =   0dB <<<
    //         2 =  -3dB
    //         1 =  -6dB
    //         0 =  -8dB
    //
    // <2:0>   PGA Gain
    //         7 =   0dB
    //         6 =  -3dB <<<
    //         5 =  -6dB
    //         4 =  -9dB
    //         3 = -15dB
    //         2 = -21dB
    //         1 = -27dB
    //         0 = -33dB
    //

    // BK4819_WriteRegister(BK4819_REG_13, 0x03BE);  // 0x03BE / 000000 11 101 11 110 /  -7dB
    // BK4819_WriteRegister(BK4819_REG_12, 0x037B);  // 0x037B / 000000 11 011 11 011 / -24dB
    // BK4819_WriteRegister(BK4819_REG_11, 0x027B);  // 0x027B / 000000 10 011 11 011 / -43dB
    // BK4819_WriteRegister(BK4819_REG_10, 0x007A);  // 0x007A / 000000 00 011 11 010 / -58dB
    // 使用BK4819_WriteRegisterGroup
    BK4819_WriteRegisterGroup_t agcGainReg[] = {
            {BK4819_REG_10, 0x007A},  // 0x007A / 000000 00 011 11 010 / -58dB
            {BK4819_REG_11, 0x027B},  // 0x027B / 000000 10 011 11 011 / -43dB
            {BK4819_REG_12, 0x037B},  // 0x037B / 000000 11 011 11 011 / -24dB
            {BK4819_REG_13, 0x03BE}   // 0x03BE / 000000 11 101 11 110 / -7dB
    };
    BK4819_WriteRegisterGroup(agcGainReg, ARRAY_SIZE(agcGainReg));

    if (amModulation) {
        // BK4819_WriteRegister(BK4819_REG_14, 0x0000);
        // BK4819_WriteRegister(BK4819_REG_49, (0 << 14) | (50 << 7) | (32 << 0));
        BK4819_WriteRegisterGroup_t agcGainReg[3] = {
                {BK4819_REG_14, 0x0000},  // 0x0000 / 000000 00 000 00 000 / -79dB
                {BK4819_REG_49, (0 << 14) | (50 << 7) | (32 << 0)} //0x1A00 / 00 1100100 0010000 / 50, 32
        };  
    } else {
        // BK4819_WriteRegister(BK4819_REG_14, 0x0019);  // 0x0019 / 000000 00 000 11 001 / -79dB
        // BK4819_WriteRegister(BK4819_REG_49, (0 << 14) | (84 << 7) | (56 << 0)); //0x2A38 / 00 1010100 0111000 / 84, 56
        BK4819_WriteRegisterGroup_t agcGainReg[3] = {
                {BK4819_REG_14, 0x0019},  // 0x0019 / 000000 00 000 11 001 / -79dB
                {BK4819_REG_49, (0 << 14) | (84 << 7) | (56 << 0)} //0x2A38 / 00 1010100 0111000 / 84, 56
        };
    }
    // BK4819_WriteRegister(BK4819_REG_7B, 0x8420);
    agcGainReg[2].Register = BK4819_REG_7B;
    agcGainReg[2].Value = 0x8420;

    BK4819_WriteRegisterGroup(agcGainReg, ARRAY_SIZE(agcGainReg));

}
```

`BK4819_WriteRegisterGroup`的实现：
```c
void BK4819_WriteRegisterGroup(BK4819_WriteRegisterGroup_t *Group, uint16_t Count) {
    for (uint16_t i = 0; i < Count; i++) {
        BK4819_WriteRegister(Group[i].Register, Group[i].Value);
    }
}
```
`BK4819_WriteRegisterGroup_t`的定义：
```h
typedef struct BK4819_WriteRegisterGroup_t {
    uint16_t Register;
    uint16_t Value;
} BK4819_WriteRegisterGroup_t;
```

使用`$(MAKE) build ENABLE_CHINESE_FULL=4 ENABLE_DOPPLER=1 ENABLE_SPECTRUM=1 ENABLE_FMRADIO=1 ENABLE_MDC1200=1 ENABLE_MDC1200_EDIT=1 ENABLE_MDC1200_CONTACT=1`编译，版本为`git checkout 0.13.2`，从占用

```
arm-none-eabi-size firmware
   text    data     bss     dec     hex filename
  61384      52    3596   65032    fe08 firmware
```

降到

```
arm-none-eabi-size firmware
   text    data     bss     dec     hex filename
  61336      52    3596   64984    fdd8 firmware
```

方法不难，但是需要耐心。也许AI可以帮忙。

xiongsp, 2025-05-20:
经过测试，应当尽量对常量部分进行折叠。含有变量的可能会影响性能。可以考虑编写llvm pass来实现。

xiongsp, 2025-05-20:
函数`enable_msg_rx`的最后几行

```c

        // clear FIFO's then enable RX
        // BK4819_WriteRegister(0x59, (1u << 15) | (1u << 14) | fsk_reg59);
        // BK4819_WriteRegister(0x59, (1u << 12) | fsk_reg59);

        // // clear interrupt flags
        // BK4819_WriteRegister(0x02, 0);

        BK4819_WriteRegisterGroup_t pgroup2[] = {
            {0x59, (1u << 15) | (1u << 14) | fsk_reg59},   // clear FIFO's
            {0x02, 0},                                   // clear interrupt flags
            {0x59, (1u << 12) | fsk_reg59},              // enable RX
        };
        BK4819_WriteRegisterGroup(pgroup2, ARRAY_SIZE(pgroup2));

//			BK4819_RX_TurnOn();

        // enable interrupts
//			BK4819_WriteRegister(0x3F, BK4819_ReadRegister(0x3F) | BK4819_REG_3F_FSK_RX_SYNC | BK4819_REG_3F_FSK_RX_FINISHED | BK4819_REG_3F_FSK_FIFO_ALMOST_FULL);
    } else {
        // BK4819_WriteRegister(0x70, 0);
        // BK4819_WriteRegister(0x58, 0);
        BK4819_WriteRegisterGroup_t pgroup[] = {
            {0x70, 0},
            {0x58, 0},
        };
        BK4819_WriteRegisterGroup(pgroup, ARRAY_SIZE(pgroup));
    }
}
```

该demo展示2处折叠。仅折叠第一处导致+4B；仅折叠第二处导致+4B；两处均折叠最终-4B。这是一些测试，供参考。最终实现可以尽量在函数内部多进行折叠。经过IDA反编译发现开启gcc的-Os参数后编译器会尽可能复用相似数据段，所以这就是多折叠常量可以带来空间性能的提升的原因。

losehu, 2025-05-20:
thx bro, 你能帮我提交pull request嘛

xiongsp, 2025-05-20:
这个可能一下子搞不定，因为整体代码很混乱，前后不一致。我趁刚才也做了下实验，方便后人能够优化。我也尽力优化后提交pr。

一次调用`BK4819_WriteRegister`耗费10B：

[image]

一次无参数调用函数（例如调用`BK4819_DisableDTMF`）耗费4B，该函数占用12B：

```c
void BK4819_DisableDTMF(void) {
    BK4819_WriteRegister(BK4819_REG_24, 0);
}
```

[image]

也就是说，如果一个WriteRegister重复了2次以上，那么将他抽象为一个函数是更好的选择（前方案耗费$10x$Byte，后者$12+4x$，简单的线性规划）

这是一个思路，因为我看到了很多重复的调用：

（3次，根据线性规划后者更优）
[image]

（有其具体实现但没有调用，浪费空间）
[image]

[image]

losehu, 2026-09-20:
已经在133固件采用

closed 2026-09-20 by losehu